armink/FreeModbus_Slave-Master-RTT-STM32 #19: 用modbus poll查询从机时，经常提示timeout err

dangyanglim, 2016-03-29:
将"slave trans",serial_soft_trans_irq,这个线程优先级从10设为0，丢包率减少了不少，但仍然有25%丢包率。
485波特率是9600，硬件是stm32，RS485的硬件不是串口。

armink, 2016-03-29:
有没有修改过底层驱动及核心源码？如果修改过，建议直接使用 GitHub 上的最新源码，再试试。

dangyanglim, 2016-03-29:
@armink 我用的rt-thread里的代码。
我现在把
    rt_timer_init(&timer, "slave timer",
                   timer_timeout_ind, /* bind timeout callback function */
                   RT_NULL,
//                   (50*usTim1Timerout50us)/(1000*1000/RT_TICK_PER_SECOND),
					10,
                   RT_TIMER_FLAG_ONE_SHOT); /* one shot */

固定为10*10毫秒。

modbus poll发送了300个包没有丢包，估计是3.5times那个时间过短。再测测。

dangyanglim, 2016-03-29:
连续发2000个包都有回复，看来是时间问题了。

armink, 2016-03-29:
我怀疑是 `RT_TICK_PER_SECOND` 跟我的源码默认不一致

dangyanglim, 2016-03-30:
@armink RT_TICK_PER_SECOND都是100的吧，我现在定时30毫秒没丢包过，9600的波特率。

armink, 2016-03-30:
`RT_TICK_PER_SECOND` 是可以修改的，我的源码中默认是 **10000** ，这里应该是问题的所在。

closed 2016-04-01 by armink