armink/FreeModbus_Slave-Master-RTT-STM32 #23: 本工程线程SystemMonitor在调用eMBMasterReqxxx函数前，必须要调用 rt_thread_delay？

ericQiang, 2017-04-13:
仓主，SystemMonitor线程在调用eMBMasterReqxxx函数前，必须要调用 rt_thread_delay吗？我在测试时，遇到的问题时，如果没有调用rt_thread_delay的话，线程就卡死了。不知道仓主有没有测试到这个问题。

仓主回答：
@ericQiang 你要等待 Modbus Poll 线程里面 MB_init 初始化成功，才能调用 modbus 相关的 API。

但目前masterPoll线程优先级最高，不存在MB_init初始化失败的情况。并且，我在初始化SystemMonitor线程时将延时设置为200.但问题依然会复现。仓主可否将你提交的工程中，SystemMonitor线程中的几个与rt_thread_delay屏蔽掉测试一下？
谢谢。

armink, 2017-04-13:
应该是因为 T35 定时器初始化还没有完成，此时发起主机请求，导致状态机混乱，最终形成死锁。T35 本身时间很短，所以之前一直没有注意到这个细节。

不过出于可靠性角度考虑，我觉得问题还是要修复的，只通过让用户延时规避问题，有点不太妥。你有没有相关建议？

ericQiang, 2018-03-29:
有个提议，在eMBMasterPoll函数中swith修改如下：
`switch ( eEvent )
{
    case EV_MASTER_READY:
    //在这里做一下标记，比如给eMBState增加一种状态ESTABLISHED
    break;
…………
}`
经测试可行。全部修改diff如下：
`diff -Naur /e/work/Learning/10.RTOS/rtthread/rt-thread-master/components/net/freemodbus/modbus/include/mb_m.h ./rt-thread-master/components/net/freemodbus/modbus/include/mb_m.h
--- /e/work/Learning/10.RTOS/rtthread/rt-thread-master/components/net/freemodbus/modbus/include/mb_m.h	2018-03-18 18:16:30.000000000 +0800
+++ ./rt-thread-master/components/net/freemodbus/modbus/include/mb_m.h	2018-03-29 11:15:39.324218700 +0800
@@ -173,6 +173,18 @@
 eMBErrorCode    eMBMasterDisable( void );
 
 /*! \ingroup modbus
+ * \brief Check the Modbus Master protocol stack has established or not.
+ *
+ * This function must be called and check the return value before calling
+ * any other functions.
+ *
+ * \return If the protocol stack has been established or not
+ *  TRUE.  the protocol stack has established
+ *  FALSE. the protocol stack hasn't established
+ */
+ BOOL            eMBMasterIsEstablished( void );
+
+/*! \ingroup modbus
  * \brief The main pooling loop of the Modbus Master protocol stack.
  *
  * This function must be called periodically. The timer interval required
diff -Naur /e/work/Learning/10.RTOS/rtthread/rt-thread-master/components/net/freemodbus/modbus/mb_m.c ./rt-thread-master/components/net/freemodbus/modbus/mb_m.c
--- /e/work/Learning/10.RTOS/rtthread/rt-thread-master/components/net/freemodbus/modbus/mb_m.c	2018-03-18 18:16:30.000000000 +0800
+++ ./rt-thread-master/components/net/freemodbus/modbus/mb_m.c	2018-03-29 11:06:18.721679600 +0800
@@ -70,6 +70,7 @@
 static enum
 {
     STATE_ENABLED,
+    STATE_ESTABLISHED,
     STATE_DISABLED,
     STATE_NOT_INITIALIZED
 } eMBState = STATE_NOT_INITIALIZED;
@@ -233,7 +234,7 @@
 {
     eMBErrorCode    eStatus;
 
-    if( eMBState == STATE_ENABLED )
+    if(( eMBState == STATE_ENABLED ) || ( eMBState == STATE_ESTABLISHED))
     {
         pvMBMasterFrameStopCur(  );
         eMBState = STATE_DISABLED;
@@ -250,6 +251,19 @@
     return eStatus;
 }
 
+BOOL
+eMBMasterIsEstablished( void )
+{
+    if(eMBState == STATE_ESTABLISHED)
+    {
+        return TRUE;
+    }
+    else
+    {
+        return FALSE;
+    }
+}
+
 eMBErrorCode
 eMBMasterPoll( void )
 {
@@ -265,7 +279,7 @@
     eMBMasterErrorEventType errorType;
 
     /* Check if the protocol stack is ready. */
-    if( eMBState != STATE_ENABLED )
+    if(( eMBState != STATE_ENABLED ) && ( eMBState != STATE_ESTABLISHED))
     {
         return MB_EILLSTATE;
     }
@@ -277,6 +291,7 @@
         switch ( eEvent )
         {
         case EV_MASTER_READY:
+            eMBState = STATE_ESTABLISHED;
             break;
 
         case EV_MASTER_FRAME_RECEIVED:
`

ericQiang, 2018-03-29:
有个提议，在eMBMasterPoll函数中swith修改如下：
`	switch ( eEvent )
	{
	case EV_MASTER_READY:
	//flag eMBMasterPoll, eg. add another state ESTABLISHED
	break;
	}`
经测试可行。全部修改diff如下：
[diff3.txt](https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/files/1858736/diff3.txt)

armink, 2018-03-29:
多个状态确实可以解决问题。本地多测试下，可以提交 pr