armink/FreeModbus_Slave-Master-RTT-STM32 #24: 请问是否有FreeRTOS版本的例程？

jiekechoo, 2017-06-05:
感谢楼主的奉献精神。
RTT版本经过修改IO口，已经跑通。
经过几天的移植，在FreeRTOS上可以编译通过，但是在运行 eMBMasterReqReadHoldingRegister(2,0,1, -1); 时就卡死了，动不了。

不知道楼主是否方便提供FreeRTOS版本的例程？万分感谢。

armink, 2017-06-05:
FreeRTOS 版本是网友移植的，还缺少一些测试跟文档，所以一直没有正式发布出来。我把相关源码发到你的 gmail 邮箱了。

jiekechoo, 2017-06-06:
非常感谢。我们移植完成，也会把相关源码、资料和文档贡献出来的。我们也是开源贡献者。

armink, 2017-06-06:
👍

dhruvelectro, 2017-06-07:
Atmink,
Can I have FreeRTOS based source code, I will also test it and contribute.

jiekechoo, 2017-06-07:
@dhruvelectro I'm testing FreeRTOS based code. I will submit pr, pls wait a while.

dhruvelectro, 2017-06-07:
@armink That is nice. Thanks for reply. I will wait.

KarrXie, 2017-08-25:
@jiekechoo @armink 可以把FreeRTOS 版本的分享给我吗？非常感谢！

armink, 2017-08-30:
@KarrXie 我分享到这里了： https://gist.github.com/armink/8558f0aa03e66dc7cedaadeccaad821e

Kevinldw, 2017-09-03:
只有event的？
btw，正在学习 ：）

armink, 2017-09-03:
@Kevinldw 移植的时候，只有 event 跟操作系统有关联

frankgp, 2017-09-06:
Can you open FreeRtos version for my referecence?

flamebox, 2018-01-06:
请问基于FreeRTOS定时器部分的移植有例程吗?其它部分已经移植好了，但是主机发命令使用超时设置无效。会一直不延时发出去，感觉是定时器部分没有移植好。

armink, 2018-01-06:
@flamebox 不太熟悉 FreeRTOS ，目前能提供的就是上面那个链接中的移植代码。

ztrx, 2018-05-03:
freertos版本能发份给我吗？谢谢
459095108@qq.com

armink, 2018-05-03:
@ztrx 我上面有分享链接的

qinhao4832, 2018-07-12:
@jiekechoo 我也是，在FreeRTOS上可以编译通过，但eMBMasterPoll();	运行第二次，这个task就会卡死。请问移植哪里还要注意吗。

LiThirteen, 2020-06-05:
@armink 关于freeRtos的有关历程，以上两个连接都不能成功跳转，希望能够发一份到我的邮箱(834409761@qq.com)，您的例程对我的帮助非常大，最近在esp8266-FreeRtos-SDK 上移植您的freemodbus主机组件。

BEIMINGLI, 2021-04-21:
如果用freertos 的软件定时器要注意xTimerStart，xTimerStop，要用xxxFromISR类型，因为这些函数是在中断中调用。而且其他的信号量和事件标志组也有在中断中调用。这点要注意，要用xxxFromISR类型函数

xiaowu-611, 2021-05-08:
@BEIMINGLI 你移植过吗？我这边也移植卡死在eMBMasterPoll();里面，你如果移植过的话可以分享下代码不？多谢！xiaowu_liu@outlook.com

imgaoyang, 2022-01-17:
@armink 那个freertos的链接我也打不开，能发送到邮箱，或者再发个链接出来吗

imgaoyang, 2022-01-17:
@jiekechoo 你移植成功了吗，开源了吗

jiekechoo, 2022-01-18:
同样的使用 FreeRTOS ，移植到 其他 硬件平台：Modbus master： https://yiqisoft.cn/blog/IoT-Gateway/187.html ，modbus slave for GD32： https://yiqisoft.cn/blog/embedded/98.html

closed 2022-01-18 by jiekechoo