armink/FreeModbus_Slave-Master-RTT-STM32 #25: 为何还有modbus地址+1和-1（usAddress--; and usRegAddress++;）？

LearyLaputa, 2017-06-28:
我觉得freemodbus官方末代版本v1.5里，如mbfunccoils.c/line161的 usRegAddress++; 属于莫名其妙的操作， 似乎并没有搞清楚设备地址与通讯地址的关系， 那个usRegAddress++;应当删除，不少人在移植从机版本的时候，把这一系列usRegAddress++;全都删除。 这个++操作应当在应用层里实现，而且不是加1，应当是加1，10001，30001，or 40001.

设备地址	                            vs Modbus通讯地址
00001~09999数字量输出	设备地址-00001
10001~19999数字量输入	设备地址-10001
30001~39999输入寄存器	设备地址-30001
40001~49999保持寄存器	设备地址-40001

armink, 2017-06-30:
我印象里 plc 默认的起始地址是 1 开始的，而还有一些 modbus 设备起始地址是  0 开始。这点在 Modbus poll 软件中也有配置。所以我觉得 port 中的 -1 应该处于这个原因。

diverger, 2018-04-15:
参考：https://en.wikipedia.org/wiki/Modbus

> It is important to make a distinction between entity number and entity address:
> 
> Entity numbers combine entity type and entity location within their description table.
> Entity address is the starting address, a 16-bit value in the data part of the Modbus frame. As such its range goes from 0 to 65,535
> In the traditional standard, numbers for those entities start with a digit, followed by a number of 4 digits in the range 1–9,999:
> 
> coils numbers start with 0 and span from 00001 to 09999,
> discrete input numbers start with 1 and span from 10001 to 19999,
> input register numbers start with 3 and span from 30001 to 39999,
> holding register numbers start with 4 and span from 40001 to 49999.
> This translates into addresses between 0 and 9,998 in data frames. For example, in order to read holding registers starting at number 40001, corresponding address in the data frame will be 0 with a function code of 3 (as seen above). For holding registers starting at number 40100, address will be 99. Etc.