armink/FreeModbus_Slave-Master-RTT-STM32 #27: 是否支持Visual Studio 编译生成dll 或则lib文件

caistd, 2017-12-16:


armink, 2017-12-16:
想在 windows 平台上用 FreeModbus ?

caistd, 2017-12-16:
是的

armink, 2017-12-16:
那只需要修改 port 下的文件就好了，记得之前有网友完成了 windows 的移植，难度应该还好。

caistd, 2017-12-16:
是不是只要将freemobus文件夹下面的所有文件，在VS新建一个项目，导入进去，然后修改port文件夹函数是这个意思吧

armink, 2017-12-17:
差不多了，你可以借鉴下里面的其他 ide 里的编译参数

caistd, 2017-12-18:
modbus_rtu master开发的时候新增了带_m的代码其它的代码有改动吗？

armink, 2017-12-19:
好像有很小的改动，你可以用比较工具对比下

caistd, 2017-12-30:
eMBErrorCode
eMBRegisterCB( UCHAR ucFunctionCode, pxMBFunctionHandler pxHandler )这个函数我没有看到里面有引用具体是做什么用的，怎么使用？能不能举个例子  谢谢

armink, 2017-12-30:
这个不是用户应用使用的 api

caistd, 2017-12-30:
那是做什么用的？好像都没看到哪里有使用到这个函数？

armink, 2017-12-31:
不是的。正如其名，它是寄存器操作的回调函数。

比如：默认的寄存器存储的数据结构是顺序存储的数组，那么操作寄存器就是操作数组，如何操作就需要在这个函数中实现。

如果你不想要这类定长且有很大限制数据结构，也可以使用链表来存储寄存器，此时就需要用户自己实现这个回调函数。也可以借鉴 EDM （https://github.com/armink/EasyDataManager） 这个项目，它期初就是专门为 FreeModbus 定制的链表存储、事件驱动的库。

armink, 2017-12-31:
许久没折腾 FreeModbus ，上面我说的含义对应的回调，跟你说的回调没有关系，请忽略。

eMBRegisterCB 是注册回调函数，跟你理解的意思一样。第二个入参
 pxHandler 是这个新增功能码对应的实现函数。你实现好意思，把它放入到第二个入参即可。这里还有个分析介绍： http://blog.sina.com.cn/s/blog_817e99610102uzjy.html

caistd, 2018-01-01:
感谢 你的主机功能好像没有看到读取保持寄存器的功能，比如功能码03

armink, 2018-01-01:
都有的

caistd, 2018-01-20:
感谢 有打算做TCP主机吗

armink, 2018-01-21:
目前没有的

caistd, 2018-01-21:
已经完成modbus tcp主机

caistd, 2018-01-24:
已经再windows 平台完成 tcp 主机移植

armink, 2018-01-25:
👍