armink/FreeModbus_Slave-Master-RTT-STM32 #28: 485收发引脚切换处理

shenrenbushen, 2018-08-17:
if (xRxEnable)
    {
        /* enable RX interrupt */
        serial->ops->control(serial, RT_DEVICE_CTRL_SET_INT, (void *)RT_DEVICE_FLAG_INT_RX);
        /* switch 485 to receive mode */
        rt_pin_write(MODBUS_SLAVE_RT_CONTROL_PIN_INDEX, PIN_LOW);
    }

在这里控制485的收发引脚被拉低，485进入接收状态，但是有可能485发送还没有完成，因为我们只是把发送的数据送入发送数据寄存器，并没有检测数据有没有发出去，程序在这直接将485置为接收状态，很可能导致最后一个字节不能发出去。

shenrenbushen, 2018-08-17:
我之前有在CSDN上下载过一个只实现了从机部分的源码，然后改了串口引脚和485收发控制引脚，程序运行有问题，然后我在将485置为接收状态之前加个1ms非OS延时（波特率使用的是115200），问题就解决了。但是延时总归不好，有没有其他解决方法，或者是不是我移植的有问题。

shenrenbushen, 2018-08-17:
我看到您在前面有提到测试过使用“串口发送完成中断（TC）”写过从机代码，能否提供一下作为参考。使用发送完成中断，就不用考虑该什么时候切换了

armink, 2018-08-17:
本仓库里的 串口 发送已经使用了 TC 标志，并且必须等待发送完成。

https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/blob/c3d21aec4337eae04f0434ecb09f2315be61a95f/BSP/src/usart.c#L121-L132

armink, 2018-08-20:
没问题话就先关闭了，后续有问题再打开

closed 2018-08-20 by armink