armink/FreeModbus_Slave-Master-RTT-STM32 #31: Pin Connection

gourav1996, 2018-11-27:
Where to connect DE and RE Pin of rs485 converter to stm32f4.
And How to configure it?

Thank you.

armink, 2018-11-27:
https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/blob/master/FreeModbus/port/rtt/portserial.c#L131

see MODBUS_SLAVE_RT_CONTROL_PIN_INDEX definition

gourav1996, 2018-11-27:
Thank you for your replay.
Just few more things, I want to get data of a meter from holding register value 40002, it's slave id is 121, and it's baudrate is 9600. Could you please suggest me where to define these?

Thank you.

armink, 2018-11-27:
baudrate can be changed by eMBMasterInit: https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/blob/master/APP/src/app_task.c#L94

the slave id and register value's change, please see eMBMasterReqReadHoldingRegister demo on app_task.c