armink/FreeModbus_Slave-Master-RTT-STM32 #32: 一个关于自动分配从机地址的问题

LeonLu8601, 2018-12-13:
对于新手来说您的代码对我熟悉modbus和freemodbus的移植有很大的帮助，先谢谢您对开源的支持与帮助，我现在遇到了一个问题，需要连接多个从机，而跟多个从机分配地址是需要多次改动代码烧录非常麻烦，现在想到可以通过识别stm32的cpuid来区分不同的从机，由主机广播来分配从机的地址，请问freemodbus中对于用户自定义的功能码如何添加，从机初始化需要做出哪些修改，能否提一下您的思路，谢谢。

armink, 2018-12-13:
从机的地址有明显的范围吗？你是想做成类似 DHCP 自动获取 IP 的功能？

LeonLu8601, 2018-12-13:
> 从机的地址有明显的范围吗？你是想做成类似 DHCP 自动获取 IP 的功能？

范围是在100以内，的确是想实现类似DHCP自动获取ip地址的功能。

armink, 2018-12-14:
现在的主机这边是有记录从机数据的缓冲区，如果范围是 100 ，由于该缓冲区的存储结构为顺序存储的数组，这样会导致很多资源浪费，建议你先评估下这种方式这种的可行性。

另外，类 DHCP 功能必须要有主机配合，可对于 modbus 的通信模型来说，主从模式将会导致从机没法主动与主机发送通信，所以感觉在 modbus 这种主从协议总线上不好实现，在 CAN 之类协议上才可能

LeonLu8601, 2018-12-14:
> 现在的主机这边是有记录从机数据的缓冲区，如果范围是 100 ，由于该缓冲区的存储结构为顺序存储的数组，这样会导致很多资源浪费，建议你先评估下这种方式这种的可行性。
> 
> 另外，类 DHCP 功能必须要有主机配合，可对于 modbus 的通信模型来说，主从模式将会导致从机没法主动与主机发送通信，所以感觉在 modbus 这种主从协议总线上不好实现，在 CAN 之类协议上才可能
那如果减少主机的数量，在主机的上位机和从机中加入自定义的功能码，点击上位机的按钮发送带有自定义功能码的广播指令，从机在接收到广播指令之后发送自己的cpuid，主机上位机根据从机发回的数据进行比较，对从机进行从机地址编号，请问是否可行？

armink, 2018-12-15:
> 从机在接收到广播指令之后发送自己的cpuid

这种是没法实现的，因为

> 可对于 modbus 的通信模型来说，主从模式将会导致从机没法主动与主机发送通信，可对于 modbus 的通信模型来说，主从模式将会导致从机没法主动与主机发送通信，