armink/FreeModbus_Slave-Master-RTT-STM32 #37: 请问是否有裸机版本的portevent的实现方法？

vefone, 2019-01-05:
非常感谢，开源这么实用的modbus master代码。

想请问一下，有实现过裸机下的portevent接口吗？我现在基于这个代码，移植了串口和定时器部分，也参考了slave的event实现方法，能够正常读写从机的寄存器。不过有一个问题：如果调用两次eMBMasterReqWriteHoldingRegister连续写两个寄存器，从modbus slave软件上只能接收到第二次写的数据。我怀疑是和event队列的实现有关系。所以想问问您有没有参考。非常感谢！

errorCode = eMBMasterReqWriteHoldingRegister(MB_SLAVE_ADDR, 0, usModbusUserData[0], 0x100);
			HAL_Delay(500);
errorCode = eMBMasterReqWriteHoldingRegister(MB_SLAVE_ADDR, 1, usModbusUserData[1], 0x100);

armink, 2019-01-05:
裸机的 portevent 接口有呀，但没有裸机的 portevent_m 主机事件移植文件，感觉你需要的好像是主机的

vefone, 2019-01-05:
是啊，我需要主机的portevent_m接口，有计划吗？

vefone, 2019-01-05:
主机的裸机portevent接口应该挺多人需要用到，大体看了一下，实现起来还是有点工作量的

armink, 2019-01-05:
暂时没有，之前想折腾过，后来放弃了，建议使用 RTOS 作为事件机制的基础

vefone, 2019-01-05:
用的mcu只有8KB的RAM,怕跑系统会比较折腾。

vefone, 2019-01-05:
下载 rt-thread3.1，发现仓主的代码在里面，components\net\freemodbus