armink/FreeModbus_Slave-Master-RTT-STM32 #38: TCP master port

GeorgeJirka, 2019-01-17:
Does anybody have tcp port for master? Because I want to try modbus with lwip,.... slave is working, and master not.

rainbowyu, 2019-12-23:
> Does anybody have tcp port for master? Because I want to try modbus with lwip,.... slave is working, and master not.

do you have the tcp port?

GeorgeJirka, 2019-12-23:
No I have not. IT IS second problém.

rainbowyu, 2019-12-23:
> No I have not. It is second problem for mě.

I found it at https://github.com/cwalter-at/freemodbus doc dir. Maybe you can find it there too.

pvyleta, 2019-12-23:
If you want to run a TCP Master, chances are you are running it on moře
advance Platform such as raspberry pi. In that case consider to use some
alternative such as Libmodbus of Pymodbus. Especially the later is super
easy to start with.

Regards
Petr

GeorgeJirka, 2020-03-01:
Hi. I have a master tcp port for cwalter-at / freemodbus. It is working fine. And now I try to implement this modbus (from armlink).

[mbm_port.zip](https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/files/4271842/mbm_port.zip)