armink/FreeModbus_Slave-Master-RTT-STM32 #46: How to use library

hasan700, 2019-06-24:
Hi
Thanks for your open source code. That’s very helpful.
I want to use Modbus/slave (ASCII and RTU) on STM32f103VET6. Actually, I’m new to ARM, and really confused. There is explanation in your code, but I don’t know how to implement the code. I should copy all of the code files to micro or just the files in “FreeModbus”? In main code of project, I should include all of the code files?
 In my project, there are ten ADC inputs to micro. The micro should read them and save every ADC channel to one register separately. After that, all of the ten registers should be transmitted over Modbus protocol to PLC. So, in the Modbus code, where should I use registers for transmission? 
Thank you very much…

armink, 2019-06-24:
hi, this project base on STM32F103X MCU. It's has IAR and MDK IDE project. Please using the IDE project for development.

hasan700, 2019-06-24:
Hi.
Yes, I'm using MDK-ARM IDE for my project. I don't know how to use this library in IDE. Should I use all of the files in this directory?
 https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32
Or I should use the files in this directory? 
https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/tree/master/FreeModbus/modbus
And how to use the files? 
sorry , my question may is very basic.

hasan700, 2019-06-24:
Hi.
Yes, I’m using MDK-ARM IDE for my project. I don’t know how to use this library in IDE. Should I use all of the files in this directory?

https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32

Or I should use the files in this directory?

https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32/tree/master/FreeModbus/modbus

And how to use the files?
Sorry, my question is very basic.

armink, 2019-06-24:
Yes, This Modbus library is base on RT-Thread RTOS plantform. So I was suggest you can using all of project file by MDK-ARM IDE.