armink/FreeModbus_Slave-Master-RTT-STM32 #47: Implementing freemodbus slave

hasan700, 2019-07-31:
Hi Mr. 
It's a long time I'm involved in implementing freemodbus on stm32 and still I couldn't do it.
My micro is stm32f103vet6 and want to use UART1 (port A9 , A10) for modbus and use CubeMX, Keil5 and freeRTOS for configuration. 
By the way, I want to use modbus slave, RTU / ASCII and just want to read holding registers from micro. 
Would you plase, tell me what should I do? 
I have followed the instructions in this website: 
http://blog.naver.com/eziya76/220970378890
But, when testing with 'Modbus Poll', the 'Time out error' occurs. 
Also, I read the 'Read me' from yours and the description inside code and parts of documements of freemodbus, but !!!!!....
This is very important for me, I really appreciate if you help me.
Thank you...

armink, 2019-08-01:
I think you can test this project on you board first. It's also has 3 IDEs project.

When it runing OK on your board. Then you can proting it to other platform.

hasan700, 2019-08-02:
> I think you can test this project on you board first. It's also has 3 IDEs project.
> 
> When it runing OK on your board. Then you can proting it to other platform.
Ok, Thank you. I'll test it, Hope it works...

hasan700, 2019-08-02:
> I think you can test this project on you board first. It's also has 3 IDEs project.
> 
> When it runing OK on your board. Then you can proting it to other platform.

Ok, Thank you. 
I'll test it, hope it works...

hasan700, 2019-08-02:
Hi
I programmed the project without any modification on my board. When testing with “Modbus poll”, the “Timeout Error” occurs.
I use UART1 for Modbus. 
Am I missing something? 
Thank you so much…

armink, 2019-08-02:
check the serial config on modbus poll

```
eMBInit(MB_RTU, 0x01, 1, 115200,  MB_PAR_EVEN);
```

hasan700, 2019-08-02:
YES! that works ....
I'm very thankful you are considering my problem.

closed 2019-08-03 by armink