armink/FreeModbus_Slave-Master-RTT-STM32 #51: How to read from slave?

emrelio, 2020-02-24:
I'm trying to read registers from the slave. But I can't read anything. With the ReqWrite function or FuncWrite function. 

How can I read registers from the slave?

Thank You!

armink, 2020-02-24:
You can ues this API 

```
eMBMasterReqErrCode eMBMasterReqReadHoldingRegister( UCHAR ucSndAddr, 
                                                     USHORT usRegAddr,
                                                     USHORT usNRegs,
                                                     LONG lTimeOut );
```

emrelio, 2020-02-24:
I'm trying with this function but I did write the function name in the first message. Sorry.

But I couldn't find the variable to read the data. I mean which variable is for reading the data.

armink, 2020-02-24:
You can find the read data on the `usMRegHoldBuf` in `FreeModbus\port\user_mb_app_m.c`

emrelio, 2020-02-27:
Thank you! I did try to read but then I realized in every request the error code is always "TimeOut". Even in the reqWrite. I'm using f407 with 168MHz clock.

armink, 2020-02-29:
May be uart driver has some problem. You can check the uart ISR for data receive.

leejean-wq, 2021-12-10:
> Thank you! I did try to read but then I realized in every request the error code is always "TimeOut". Even in the reqWrite. I'm using f407 with 168MHz clock.

me too!Have you solved the problem yet